nordic-auko/nRF5-ble-timesync-demo #2: conflicting types of '__putchar()'

pescara, 2019-10-11:
Hi
I downloaded your code and tryied to build it in Segger Embedded Studio. But there the error comes:
"conflicting types of '__putchar()'

(1) you declare it in stdio.h as follow:
int __putchar(int, __printf_tag_ptr);

(2) then in rectarget.c it is defined as following:
int __putchar(int ch, FILE * p_file)
{
    UNUSED_PARAMETER(p_file);
    UNUSED_VARIABLE(app_uart_put((uint8_t)ch));
    return ch;
}
Can you help me?
Is is ok if I just uncomment the (2) declaration? Will it still run? I'm very very pleased for any help since my thesis will highly depend on it... : (

nordic-auko, 2019-10-30:
Hello,
it looks like this file has been updated in more recent nRF5 SDK versions.
Can you try the attached retarget.c (copied from [nRF5 SDK 16.0.0](http://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v16.x.x/))?

[retarget.zip](https://github.com/nordic-auko/nRF5-ble-timesync-demo/files/3787869/retarget.zip)

m3x1m0m, 2019-12-03:
@nordic-auko Tested it and it seems like a functioning workaround. Can you update it? Should I generate a PR? Also I think the project should embed a .gitignore like the following
[gitignore.zip](https://github.com/nordic-auko/nRF5-ble-timesync-demo/files/3917683/gitignore.zip)

nordic-auko, 2019-12-13:
Thanks, @m3x1m0m ! 
I plan to bump this repo to the latest SDK version sooner rather than later anyway, so there's no need for a PR. 
And thanks for the tip regarding .gitignore!

nordic-auko, 2020-03-09:
Added SDK 16 examples

closed 2020-03-09 by nordic-auko