nordic-auko/nRF5-ble-timesync-demo #6: Device sometimes couldn't receive sync_pkt sent

rrandyzh, 2020-06-15:
Hi Audun,

Have a good day.
I tested this timesync-demo under SDK-16.0 and found that sometimes the device couldn't receive any sync packet sent from master（More specifically time sync stopped itself, a system reset could resume).  However the BLE activity works well when the time sync is stopped. 
Is there any methods to debug this issue? I tried to re-init the time-sync functions but failed.
If this issue occur  how could we to resume the time-sync without system reset or SOFTDEVICE reset?

Appreciate your quick response.

Kindest Regards，
Randy

nordic-auko, 2020-06-15:
Hi Randy,

there are several things that could cause the error you see, I think. Let's try some steps to narrow down the issue.

First off, how do you monitor that the time sync is working? Is it via the GPIO toggling, calling the timestamp function, or something else?

If the GPIO has stopped toggling, it could be related to for example:
- Some other part of the application has reconfigured the required hardware peripherals (TIMER, PPI, GPIO, EGU).
- There is a bug in the time sync code that has caused one of the aforementioned hardware peripherals to be disabled or misconfigured

If the GPIO keeps toggling but you can see it is drifting out of sync, it could be for example:
- Softdevice timeslots are no longer granted. Check to see if the radio_callback() function in time_sync.c is triggered after things have stopped working.
- The timing master is no longer transmitting sync packets (also check if radio_callback() on master side is triggered)

Best regards,
Audun

rrandyzh, 2020-06-17:
Hi Audun,

Many thanks for your reply.:-)

The issue was fixed now.  The 'Optimization Level'  of SES result in this failure. If configure option 'Optimization Level' with 'LEVEL 2 for size', the device could reproduce the sync drift(sync packet loss) within an hour. Now I configured this option with 'LEVEL 0 no optimization', the demo code runs as expected for  12 hours.

Kindest Regards，
Randy

closed 2020-06-17 by rrandyzh