nordic-auko/nRF5-ble-timesync-demo #17: Usage with BLE LR (Coded Phy)

mkunzecw, 2021-08-16:
We tried today to modify the example to use BLE LR 500 or 250kbits

```
 NRF_RADIO->MODE     = RADIO_MODE_MODE_Nrf_2Mbit       << RADIO_MODE_MODE_Pos;
```

we tried:

BLE 1MBit worked
```
 NRF_RADIO->MODE     = RADIO_MODE_MODE_Ble_1Mbit       << RADIO_MODE_MODE_Pos;
```

neither 500 or 250Kbit worked
```
 NRF_RADIO->MODE     = RADIO_MODE_MODE_Ble_LR500Kbit       << RADIO_MODE_MODE_Pos;
```
```
 NRF_RADIO->MODE     = RADIO_MODE_MODE_Ble_LR250Kbit       << RADIO_MODE_MODE_Pos;
```

we also tried to increase timeslot length and extension
```
#define TS_LEN_US                            (20000UL)
#define TX_LEN_EXTENSION_US                  (20000UL)
```

we also tried to to reduce the frequency from 30Hz to 5Hz

It seems that the Sender is sending (its toggling the output) but the receiver is not entering the IRQ Handler (nor toggling).

Maybe you have an idea?

mdreyercw, 2021-08-30:
In a next step we tried to ajust the radio parameter, especially the packet fromat.
The changes we made can be found in my repository.

[https://github.com/mdreyercw/nRF5-ble-timesync-demo](url)

From the current consumtion of the transmitting board, it looks like it is sending with 125kBit.
Therefore we are thinking the other bord is unable to recieve or interpret the message.

nordic-auko, 2021-09-02:
Hi @mdreyercw ,
sorry for the slow response to this one.. 

I've not tried this radio mode myself, but I do have some thoughts.

Did you change any other NRF_RADIO registers in addition to MODE?
The "BLE Long Range" modes has a slightly different packet layout compared to the 2Mbit and 1Mbit modes, and some other register values has to be adjusted accordingly. See here for details: https://infocenter.nordicsemi.com/topic/ps_nrf52833/radio.html?cp=4_1_0_5_17_0#concept_hz4_cjl_4r 

The [NRF_RADIO->PCNF0](https://infocenter.nordicsemi.com/topic/ps_nrf52833/radio.html?cp=4_1_0_5_17_14_51#register.PCNF0) register should be changed. I believe the PLEN field must be set to value 3 (LongRange), the CILEN field to value 2, and the TERMLEN field to value 3. I'm not entirely sure if you can adjust these values according to your specific needs.

When using the long range modes, the timing will also be off. You will need to adjust TX_CHAIN_DELAY_PRESCALER_0 to account for the difference in transmission time.

mkunzecw, 2021-09-07:
Hey,

we got it working with heavily modified code from the DTM example now (different BA, dynamic LEN and S0 and some other parameters). But we need to refine it a bit if you are interested in a solution. 

We also found that the phase of the 2 square waves is sometimes 90° out of phase (perfectly) and is jumping between 0° and 90° any idea?

closed 2022-07-12 by nordic-auko