nordic-auko/nRF5-ble-timesync-demo #19: Maybe bugged branch?

David140, 2021-12-22:
Hi all!
I'm full time working to update my code with the new branch (count_revamp).
I download the branch, flashed my boards and press the button to start the syncronization.
This is the situation after some seconds:
[image]

Is it a bug?
I think I will need a lot of support in the following weeks.
Thanks for the help

crazyskady, 2021-12-23:
Hi, David
I happened to use this demo a few months ago. The demo works well, thanks to the authors for this excellent feature.
Maybe some suggestions could help you to find the root cause.
1. Ensure the sender and receiver using the same RF address and channel. (If you can sync at start, I guess this tip is okay in your environment)2. Ensure there's only one sender using the same RF address and channel over the air, if you have more than one sender, reminder to update the address & channel. (This could be possible, I meet similar issue in my environment.)3. Using same module (both 52832 or both 52840 is verified in my test) and same compile environment for download the application for both sender&receiver. 

--------------------------------

BR,
skady

抄送人：Subscribed ***@***.***>
主题：[nordic-auko/nRF5-ble-timesync-demo] Maybe bugged branch? (Issue #19)
日期：2021年12月23日 06点33分

David140, 2021-12-23:
Hi crazy, thanks for the answer!

> 1. Ensure the sender and receiver using the same RF address and channel.

I think this is ok, the two boards can sync at the beginning, just for like 30s.

> 2. Ensure there's only one sender using the same RF address and channel over the air, if you have more than one sender, reminder to update the address & channel.

I'm using at the moment just one sender (that is the BLE central) and one receiver (that is the BLE peripheral).

> 3. Using same module

I'm using a NRF52840DK as BLE central (and sync packets sender) and a Adafruit Feather NRF52840 as BLE peripheral (and sync packets receiver). The MCU is the same (NRF52840) so I don't think this could be a problem. Maybe I can try to use 2 NRF52840 dongles (to exploit 2 identical boards) but I would be surprised if something will change.

crazyskady, 2021-12-30:
Hi, David

I guess different board have no impact if use same module, not sure. But I guess crystal oscillator configurations (PPM and so on) in sdk_config.h could have some impact, not sure, you can set them same in both. :)May be you can send email to author ask for help directly. I can't remember no more details due to I have left the company which use this module. :)
--------------------------------

BR,
skady

抄送人：Crazy Skady ***@***.***>,  Comment ***@***.***>
日期：2021年12月24日 01点44分

nordic-auko, 2022-01-07:
Hi David,

I tested the revamp branch again and see indeed the same issues as you.
Will need to spend more time on this to understand if the issue was always there, or if there was a local difference that did not make it into the commit.

Specifically it seems that the 2nd timer (in counter mode) is not handled properly, as the offset is always in time units of TIME_SYNC_TIMER_MAX_VAL / 16000000 seconds.

David140, 2022-03-05:
Hello.
Another month without an answer.
We are developing a new product and this branch is still bugged.
Could we have the correct commit in few days please?
Thanks.

cperezmicrofactorycoop, 2022-03-10:
Hi, I'm also hoping a fix on this situation would occur as I'm getting similar issue.

David140, 2022-03-11:
Hi @cperezmicrofactorycoop 
Audun has just pushed an errata corrige commit on the counter_revamp branch.
Thanks again!

nordic-auko, 2022-07-12:
Closing after inactivity

closed 2022-07-12 by nordic-auko