nordic-auko/nRF5-ble-timesync-demo #20: Build error: undefined reference to `nrfx_ppi_channel_alloc'

Nyovelt, 2022-01-17:
[image]

I have read #12 and enabled `NRFX_PPI_ENABLED` as shown in the screen, but the IDE still consider it as error.

David140, 2022-01-18:
Hi Nyovelt,
I needed to make these changes to build the project:
    1. SDK_CONFIG: #define PPI_ENABLED 1 (it was 0)
    2. INSERTING: nrfx_ppi.c file inside the project
I think at the moment the branch is still bugged.

quannguyen123456, 2022-01-21:
hi Mr David140
I did all 1 and 2, but still error
Do you have some ideas?
thank you very much.

David140, 2022-01-22:
Hi,
> I did all 1 and 2, but still error
No, sorry, I don't know, I just did these steps and the code compiled.
But anyway the new branch is still bugged, I'm waiting the support of @nordic-auko.
Thanks

Nyovelt, 2022-01-26:
I have switch to branch `sdk16` and worked well.

closed 2022-01-26 by Nyovelt