nordic-auko/nRF5-ble-timesync-demo #21: Porting the project to the NRF52DK?

Fdi-Inno, 2022-05-09:
Hi, first of all thank you for the great work that has been done here. Unfortunately I don't posses a nRF52840-DK.I was considering porting the project to the NRF52DK, would it be easy? Wouldn't it take too long?

Thank you in advance!

nordic-auko, 2022-05-09:
Hello, the code runs fine on both nRF52-DK, and nRF52840-DK :)
Use "pca10040" Makefile for nRF52-DK

Fdi-Inno, 2022-05-10:
Hi, thank you for the quick answer, I figured it out but couldn't delete my message.  

I got a problem though, when opening the project with Keil, a prompt tells me that the DefiveFamilyPack 8.27.0 is required.
It directs me to the pack installer, which can't find the pack. As it is in the list, I installed it myself in the pack installer, but there seems to be a link problem.

Could it be because i only got SDK 17.1.0 installed?

I only found this solution, on google, it feels like trying it made things worse :
https://devzone.nordicsemi.com/f/nordic-q-a/55378/mdk-uvision-sdk-16-0-keil-family-pack-installation/224413#224413

nordic-auko, 2022-05-10:
Is it an option to use GCC? I've not tested this code in Keil for a long time..

There is a Pack Installer in Keil where you can search for and install missing packs: https://www.keil.com/support/man/docs/uv4/uv4_ca_packinstaller.htm

All the Nordic-produced packs are also available here: https://developer.nordicsemi.com/nRF5_SDK/pieces/nRF_DeviceFamilyPack/