nordic-auko/nRF5-ble-timesync-demo #30: How to reduce power consumption in receiving mode

shekmun, 2023-08-25:
I've test the power consumption both in receiving and sending mode on the nrf52832 minimum system board.
In receiving mode, the average current is about 13mA. But after switched to sending mode, the average current reduced to 1mA.
Which means that the "receiving" (something like scan) costs about 12mA. So is it possible to reduce power consumption in "receiving" mode?

nordic-auko, 2023-11-21:
The system is not optimized with regards to power consumption. That is, the transmitter is somewhat optimized in the sense that it only uses the radio when it actively transmits. The receiver on the other hand, keeps the radio in receive mode for as long as possible. To improve this the receiver needs to know when to expect packets to arrive, and only keep the radio on accordingly.

shekmun, 2024-01-02:
Thanks for your advising. But can I know which parameters can I change in the code to make it receive in shorter time (I can accept if the sync speed is slower and maybe not as long as possible), or only receive the packet at the exact moment with very short period (maybe this needs being synced). I've tried something like "m_timeslot_req_earliest" and "m_timeslot_req_normal", but no success.

rdebrum-cdi, 2024-04-19:
Any further updates on this? I too see more than double the current draw in receiving mode and for my customer's sake will need to bring this down.
Do you mean the firmware should not try to extend the MPSL? If the transmit frequency is known, how do you think it is possible that we could optimize the receiver to only be in receive mode for the expected window of time  (on the general tick of the freewheel)?