nordic-auko/nRF5-ble-timesync-demo #37: MPSL Assert

MB-Conti, 2025-01-15:
Hi,
i run a slightly adapted version of the timesync demo and experience sometimes a freezing of a time "slave" (mpsl assert 80, 2).

From my tests my assumption is, that this is due to the scheduling of a new earliest MPSL request if the slot was already extended up to 128 seconds [Line 454](https://github.com/nordic-auko/nRF5-ble-timesync-demo/blob/1bab6633a6155e29086d62162b68076d73f7a4fa/ncs/timesync_sample/src/timesync/time_sync_nrf52.c#L454).

If running into this path a new earliest timeslot request is triggered. However, the timeslot_end_handler() might not be called. As the MPSL_TIMESLOT_SIGNAL_ACTION_REQUEST will close a current timeslot and a new one is planned. My assumption is that the assert is then triggered due to improper closing of the timeslot. 

From my perspective the condition might not be necessary. If the Extension Timer triggers and we are above the specified extended time, it might be possible to just skip the extension request and let the "time is about to end" timer handle the closing/rescheduling of the timeslot accordingly.

Due to time constraints I was not able to follow this up completely. However, I thought raising this issue might help in the future.

nordic-auko, 2025-02-05:
Thanks for the info and your analysis!
Which version of NCS are you using in your adapted version? I'd like to look into this assert

MB-Conti, 2025-02-05:
I observed the behaviour in nrf Connect v2.7

nordic-auko, 2025-02-05:
This MPSL assert happens when a radio IRQ is triggered outside of a timeslot. So I think your assessment is reasonable, and that the proper cleanup is not done upon the end of the timeslot.